arduino-libraries/Arduino_AdvancedAnalog #39: Number of ADC channels arbitrarily limited

tfry-git, 2023-04-13:
These lines of code:

https://github.com/arduino-libraries/Arduino_AdvancedAnalog/blob/d989bcf56fdf5d74b5fb9d076f57eca2a44771a8/src/HALConfig.cpp#L166

and

https://github.com/arduino-libraries/Arduino_AdvancedAnalog/blob/d989bcf56fdf5d74b5fb9d076f57eca2a44771a8/src/AdvancedAnalog.h#L38

(also referenced, here: https://github.com/arduino-libraries/Arduino_AdvancedAnalog/blob/d989bcf56fdf5d74b5fb9d076f57eca2a44771a8/src/AdvancedADC.h#L39)

limit multi-channel sampling to a sequence of max length 5. The STM32H7 actually supports 16 channels (accordingly #defines up to ```ADC_REGULAR_RANK_16``` exist), and arguably the feature is most useful when more than a handful of pins are involved.

Now, I am not quite sure what is the long term plan for cross-hardware support in this library. Depending on that, the proper fix will either be a plain adjustment of the values, or a small mess of #ifdefs.

iabdalkader, 2024-02-14:
Sorry for taking so long to reply to this issue, I haven't been watching the repo for a while as I thought it was maintained by someone else. Re your issue, I can't remember exactly why I limited the channels to that specific number, it's probably because we only broke out a few ADC channels, this was before the GIGA. Can you point out where in the datasheet it mentions the maximum number of channels that can be sampled successively ? If they are indeed 16, there's no reason not increase them.

tfry-git, 2024-02-14:
Pinpointing the documentation is complicated by the fact that STM provides separate huge documents for each processor variant within the H7 family. Anyway, for the H747 of the giga that would be ST RM0399 version 4.0, section 26.4.11 (page 970), and the detail descriptions sections 26.6.11 through 26.6.15 on the ADC_SQRx registers.

For determining that this is indeed a common spec of the H7-family, I guess the more accessible reference is the HAL driver: targets/TARGET_STM/TARGET_STM32H7/STM32Cube_FW/STM32H7xx_HAL_Driver/stm32h7xx_ll_adc.h lists 16 ADC_REGULAR_RANK_XY ranks, without any #if for processor variant.

iabdalkader, 2024-02-22:
@tfry-git Thanks for the references. I have this on my todo list and will fix it as soon as I can.

PR #61 (2024-03-20): AdvancedADC: Increase the maximum number of channels.
Fixes #39
closed 2024-03-20 by iabdalkader