arduino-libraries/Arduino_AdvancedAnalog #46: DAC maximum Voltage is low (0.33V)

hzarei4, 2023-10-29:
Hello,

Using the Arduino Giga R1 WiFi and the pin DAC0, I tried to generate an analog signal but the maximum Voltage of this signal is ~330 mV. I used the function AdvancedDAC.write() and a sample code provided in this link and set the value as the maximum of the resolution(for example 255 for 8-bit resolution):

[link from the arduino tutorial](https://docs.arduino.cc/tutorials/giga-r1-wifi/giga-audio?_gl=1*16322yn*_ga*MTMyNTY5MzIyMS4xNjk4NDExNjkw*_ga_NEXN8H46L5*MTY5ODU4NjcwMS4yLjEuMTY5ODU4OTkzNC4wLjAuMA..)

Is it possible to have an analog output of 3.3V from the Arduino Giga R1 WiFi?

facchinm, 2023-10-30:
Hi @hzarei4 ,
I just tested on a Giga and everything looks fine (the swing is 0-3.3V). Do you have any load connected to the DAC pin directly? Alternatively, 0.33V looks suspiciously like a 10x attenuation on the tool you are using to measure, can you double check by measuring 5V pin that the tool is ok?

hzarei4, 2023-10-30:
I have connected the Arduino to the PC, and when I try to connect the 5V pin, the Windows says that the Voltage can not be provided.
Also, I connected the DAC0 to the oscilloscope and there is no other load on the system.

hzarei4, 2023-10-30:
@facchinm It is strange, I tried with 3.3V and 5V output and the oscilloscope shows 0.3V and 0.5V values.

closed 2023-10-30 by per1234

hzarei4, 2023-10-30:
Figured it out that there was a problem with the measurement. 
Thanks a lot! :)