arduino-libraries/Arduino_AdvancedAnalog #51: Sampling with very high frequencies did not work as expected.

divebubble, 2023-12-18:
First of all, thank you for this great library.

Sampling with very high frequencies did not work as expected.

According to the documentation for the STM32H7, sampling rates of > 2 MHz should be possible. I have read that you are working with a slow timer.

In my case I need at least 4 ADC channels with a sampling rate of at least 8bit / 600'000Hz each. The problem is that I should work with ultrasonic receivers that receive at a frequency of 200KHz. In order to analyze the received data correctly, the triple sampling rate is the lower limit.

So far I have managed to sample at 70KHz per channel, distributed over two ADCs with 2 channels each. As soon as I go higher (80KHz) the program no longer starts.

Is this due to this "slow timer" or are my requirements too high?
It's been about 30 years since I wrote anything in C++ or assembler, but with a few hints I might be able to adapt the code myself.

leonardocavagnis, 2023-12-19:
Hi @divebubble,
Please, could you share your sketch code?

divebubble, 2023-12-19:
Hi @leonardocavagnis 

Arduino Uno R1, Arduino_AdvancedAnalog 1.3.0
The test sketch runs sometimes up to a sampling rate of 974'800Hz per channel. 
Sometimes it stops already at 450'000 or already at 70'000

renamed sketch to .txt as of upload restrictions
[TestingFrequency.ino.txt](https://github.com/arduino-libraries/Arduino_AdvancedAnalog/files/13714791/TestingFrequency.ino.txt)

iabdalkader, 2023-12-19:
> Is this due to this "slow timer" or are my requirements too high?

I'm not sure which slow timer you're referring to, the ADC trigger timer frequency is not fixed as it gets set to the `sample_rate` argument you pass to `begin()`. The ADC is clocked at 64 MHz, that's fixed, but with this clock the total conversion time is (Sample Time + Resolution/2 + 0.5) which is 17 cycles (or 0.265625 uS) so it should max out at ~3.6 Msps in theory, if everything else can keep up, in reality that's not likely.

Thanks for posting the sketch, I can test it later.

iabdalkader, 2023-12-19:
>> Arduino Uno R1, Arduino_AdvancedAnalog 1.3.0

@divebubble This library only works on H7 micro-controllers, I think maybe you meant Giga or Portenta ?

iabdalkader, 2023-12-19:
@divebubble 

I tested your sketch, and noticed this:
```
      if (adc_1.available()) {

        SampleBuffer buf1 = adc_1.read();
        SampleBuffer buf2 = adc_2.read();
```

`read()` is blocking, you only test if ADC1 has buffers, but you don't test ADC2, so it may block forever.

iabdalkader, 2023-12-19:
@divebubble This should fix the issue, I've been running your sketch for a while without any lockups. Are you able to test this PR ?

https://github.com/arduino-libraries/Arduino_AdvancedAnalog/pull/52

divebubble, 2023-12-19:
@iabdalkader 
**regarding the slow timer**... some time ago there was a note about Known issues. But this has been removed in the meantime
ADC is running at the slowest possible clock (PCLK), should probably set up a PLL and change the clock source.

**Arduino Uno R1**
Sure, I just mistyped. It's an Arduino Giga R1 WiFi

**adc_2.available()**
I have just one signal generator, so i can check only one.

I check PR #52 and let you know.

iabdalkader, 2023-12-19:
>>  ADC is running at the slowest possible clock (PCLK), should probably set up a PLL and change the clock source.

This is still true, but even at the slowest clock, it should be able to handle a very high sample rate, I think 600KHz is is doable, but let me know if it doesn't work.

>>  I have just one signal generator, so i can check only one.

If you use more than one ADC (like in your example sketch) you must poll each instance with `available()` before calling `read()` on that instance, otherwise `read()` can block. If that's okay with you, to block on `read()`, then don't have to worry about polling.

iabdalkader, 2023-12-19:
I can try use the microseconds timer, if it works, to set the timestamp, but moving it to another IRQ handler might be more tricky. The only difference will be the timestamp will be set before the transfer is done vs after.

divebubble, 2023-12-20:
Donation done... Thanks for you work.
Whether the timestamp is set before or after the transfer is not important. But microseconds or the cycle count of the processor would be better for synchronizing the different streams. If not, I can also try to calculate the time based on the audio waves.

iabdalkader, 2023-12-20:
I switched to us ticks for the timestamp in the same PR.

closed 2023-12-21 by iabdalkader