arduino-libraries/Arduino_AdvancedAnalog #54: Synchronized ADC Capture

jmdodd95682, 2024-02-04:
The AdvancedADC gives you two options for capturing two analog signals simultaneously.  The first is to bind both analog input pins to the same AdvancedADC instance.  This works great, but sadly, means your sample rate is cut in half.  The sample_rate provided applies to  the sampling of both inputs.  So a 1M Samp./sec rate really means each is sampled at 500K Samp./sec.   If you trying to push the limits of sample rate, this is not a great solution.

The second way is to bind each analog input to a different AdvancedADC.  The library will automatically assign the two pins to two different ADC blocks.  The sample rate of each will be maximum.  Very nice.  However, this creates another problem.  The begin() routine takes approximately 100us on GIGA R1 to initialize and start the ADC capture.  So if I say something like:

               
`adc_input[0].load(res, sample_rate, num_samples, QUEUE_DEPTH,1,&(probes[0]));                
adc_input[1].load(res, sample_rate, num_samples, QUEUE_DEPTH,1,&(probes[1])));`

What happens is that the second ADC input capture is delayed by about 100us.  So the two signals are not synchronized.  If you're trying to capture two signals with reasonable synchronization, that is not going to work.  I analyzed the timing of each line of code in the begin() routine and found that two lines are causing almost all the delay.

`descr->pool = new DMABufferPool<Sample>(n_samples, n_channels, n_buffers); //40us`

and

`if (hal_adc_config(&descr->adc, ADC_RES_LUT[resolution], descr->tim_trig, adc_pins, n_channels) < 0) {
        return 0; //60us
    }`

I have a solution.  I created two new routines in AdvancedADC.cpp  I call load() and fire().  
load() is identical to the begin() routine up to a point, including the long-delay lines above.  It then returns.  Essentially
load() allocates the DMA and ADC etc, but does not start the capture.  The fire() routine (which only takes about 4us) does the rest of
what was in begin() and starts the capture.  This is working and providing a synchronization error of about 4-6us.  Not bad. 

This is probably NOT the solution that someone else would prefer, but it works for me.  Now I have full sample-rate and okay synchronization between two or three analog inputs.

I guess I would say  the issue I have is that begin() takes too long (106us!).  Either this needs to be shortened significantly or the idea of the load() and fire() would also work.

leonardocavagnis, 2024-02-05:
Dear @jmdodd95682,
Thanks for the useful analysis and measurements! 
Could you please create a PR with your proposed solution? I'll take a look at it.
Thanks

jmdodd95682, 2024-02-06:
Sure.  I can do this.

jmdodd95682, 2024-02-06:
okay I submitted a PR.  I renamed the functions to ready() and start().  I also added a function called clear() to the AdvancedADC.  I found cases where when starting the ADC capture the DMA would (for some unknown reason) think it already had DMA Buffers in the read queue.  Running clear() after ready() and before start() fixes this issue.

jmdodd95682, 2024-02-07:
I had another thought.  Rather than getting rid of the existing begin() we should just keep it for backward compatibility with people's existing code.  Instead, add a new bool flag to begin with a default of false.  When this flag is set true, it does all the setup, but does not start.  If the flag is false, it does the legacy behavior of doing the setup and start.  

To avoid duplicate code, the begin() will call the new start() routine when the flag is false.  So, we get full backward compatibility and get the new functionality with no duplicated code.

PR #65 (2024-03-25): Add support for dual ADC mode.
Fixes #54
Fixes #59
closed 2024-03-25 by iabdalkader