arduino-libraries/Arduino_AdvancedAnalog #68: Adding Interleaved Sampling for 2X Sample Rate improvement

jmdodd95682, 2024-03-27:
Looking through the Reference Manual for the STM32H7xx I see there is another Dual mode called Interleaved.  It seems likely that by putting an input signal into two Analog Input pins (e.g. A0 and A1) and enabling Dual interleaved mode it will be possible to sample at twice the current maximum rate.  So, instead of say 2.9MS/sec we can reach 5.8MS/sec at 16b resolution.

Also, one can also enable a single DMA to handle this dual interleaved data so that all the captured data needs to re-arranging once retrieved from the DMA buffers.

I'm going to try some experiments and see if it works.

iabdalkader, 2024-03-28:
> Also, one can also enable a single DMA to handle this dual interleaved data so that all the captured data needs to re-arranging once retrieved from the DMA buffers.

Yes a single DMA channel can be used in all dual modes, to read the combined sample register. However, I read on the forums that this may be broken, the two halves of the register go out of sync, nut more importantly, I don't think we'll be able to reuse the code that we have. So I think this is secondary, not very high priority improvement.

> I see there is another Dual mode called Interleaved.

You can add a mode argument to ADCDual, and implement more modes if you like.

jmdodd95682, 2024-03-28:
Good to know about the DMA. I'll check it out.  If not then there's a little reassembly to do after capture, but it's not bad.

Yes I was thinking to add this feature to AdvancedADCDual.

Yes a single DMA channel can be used in all dual modes, to read the combined sample register. However, I read on the forums that this may be broken, the two halves of the register go out of sync, and more importantly, I don't think we'll be able to reuse the code that we have. So I think this is secondary, not very high priority improvement.

jmdodd95682, 2024-03-29:
I have the 2x sample rate using interleaved Dual mode working.  I can now sample up to 9.2MS/sec at 8b resolution (5.8MS/sec @ 16b res).  I want to do more testing before I do a pull request.  I would also like to test the unified DMA which, as you said, probably will not work reliably.

Very cool though.  I should be able to make a pretty good capture scope app with this.

iabdalkader, 2024-03-30:
Thanks @jmdodd95682 excellent work!