arduino-libraries/Arduino_AdvancedAnalog #81: [Bug] Pins from A8 to A11 cannot be used on Arduino Giga R1

mzap-bit, 2024-10-01:
Hello guys,

I am struggling to acquire a signal using A8,A9,A10, A11 pins of Arduino Giga R1.

Here's the modified example code I am using:
``` C
#include <Arduino_AdvancedAnalog.h>

AdvancedADC adc(A8.get()); // I must use this method to avoid compiler error
// AdvancedADC adc(PC_2C); // this doesn't wok as well I assume AdvancedAnalog relies on pins_arduino.h definitions (where the PureAnalogPins are not defined)
uint64_t last_millis = 0;

void setup() {
    Serial.begin(9600);

    // Resolution, sample rate, number of samples per channel, queue depth.
    if (!adc.begin(AN_RESOLUTION_16, 16000, 32, 128)) {
        Serial.println("Failed to start analog acquisition!");
        while (1);
    }
}

void loop() {
    if (adc.available()) {
        SampleBuffer buf = adc.read();
        // Process the buffer.
        if ((millis() - last_millis) > 20) {
          Serial.println(buf[0]);   // Sample from first channel
          Serial.println(buf[1]);   // Sample from second channel
          last_millis = millis();
        }
        // Release the buffer to return it to the pool.
        buf.release();
    }
}
```
The code compiles, but the serial values are showing values as if the pin is floating (despite I have placed a 1k resistor to 3.3V).

leonardocavagnis, 2024-10-01:
Hi @mzap-bit,

Currently, the library does not support `PureAnalogPin` on the Arduino GIGA R1 (A8, A9, A10, and A11). 
We will address this issue in the next release.

Thank you for reporting it.

mzap-bit, 2024-10-01:
Hi @leonardocavagnis ,

You're welcome! Thank you for your great work! :)

Just a small hint for the API doc: currently it is mentioned the Pins from A8 to A11 are actually supported, I've spent some time wondering what was wrong with my code trusting this info! A slight update might help other users!

BenLesArd, 2024-12-21:
Hi,
Thank you for your work.
Do you have an idea when you might work on this topic?
I also spend some hours before reading this post.
Same issue with my project, I need all the Analog ADC input for my project. 
This is not to stress you.
May I help you? I am retired and have time to spend on this.
Does adding A8, A9, A10, A11 define u value might help (these are missing in the pin description .h) ?
Merry Xmas,
Ben

BenLesArd, 2024-12-24:
> Hi @BenLesArd <https://github.com/BenLesArd>,
>
> Unfortunately, solving this issue isn’t as simple as adding the missing
> define statements.
> The core of the problem lies in the way the AdvancedADC class is
> designed. Specifically, its constructor
> would need to be modified to handle PureAnalogPin types.
>
> The definitions for these pins (A8, A9, A10, A11) are found in the GIGA
> variant of the ArduinoCore-mbed
> <https://github.com/arduino/ArduinoCore-mbed>, specifically in
> pure_analog_pins.cpp
>
> One straightforward approach would be to create a dedicated constructor
> that handles only PureAnalogPin types. However, this solution has its
> drawbacks because it wouldn’t allow a mix of “normal” pins and
> PureAnalogPin types during initialization.
>
> We won’t be able to work on this in the short term, but your help would be
> greatly appreciated!
> If you’re interested in contributing, feel free to try this approach as a
> starting point or propose any other feasible solution. 🙂
>
> Thank you for your interest and Merry Christmas! 🎄

Hi Leonardo,

Thank you for your kind answer.

I have got a look in the source code and understand better what is the
issue.
I still have a lot to learn.

Effectively, we could start with a dedicated constructor for PureAnalogPin
but the restriction of PureAnalogPin only.
It is a good starting point.

I still have to read and understand analogPinToPinName(p). Do you know
where can I find it?

I am confused with this:
            int begin(uint32_t resolution, uint32_t sample_rate, size_t
n_samples,
                  size_t n_buffers, size_t n_pins, pin_size_t *pins, bool
start=true,
                  adc_sample_time_t sample_time=AN_ADC_SAMPLETIME_8_5) {
            if (n_pins > AN_MAX_ADC_CHANNELS) {
                n_pins = AN_MAX_ADC_CHANNELS;
            }
            for (size_t i = 0; i < n_pins; ++i) {
                adc_pins[i] = analogPinToPinName(pins[i]);
            }

            n_channels = n_pins;
            return begin(resolution, sample_rate, n_samples, n_buffers,
start, sample_time);
        }
I don't think this is in the documentation of the library.
Is it correct to say that the purpose is to create an instance of
AdvanceADC without parameters and then use this form of begin to add the
AnalogPin?
Why not create a begin method with a list of PureAnalog to add?
What do you think of a method to add PureAnalogPin to an instance of
AdvanceADC (not yet begun)?
This is not clean but may be easier?

It is clear there is a lot to learn. Are you available for some exchange?

Merry Xmas.

Ben

BenLesArd, 2024-12-26:
Hi Leonardo,

I had some time to go deeper in the code.

I have found  analogPinToPinName (p) in mbed.
(ArduinoCore-mbed/variants/GIGA/pins_arduino.h)

I have read PureAnalogPin which are defined in
mbed/variants/GIGA/pure_analog_pins.h and declared
in../pure_analog_pins.cpp.
I also found that in mbed, we have 2 version of functions like AnalogRead:
one for the classic Analog Pin and one for PureAnalogPin. (in mbed/...
/pure_analog_pins.cpp)
At the end to read the adc, these functions use new mbed::AnalogIn(pin)
were pin is type of PinName.

So far, I think that it would be better to be able to accept the both type
of Analog Pin in the list of the constructor of AdvanceADC.
To do that I think the best would be to create a class for the al the
Analog Pin (I.e. AdvanceAnalogPin...). That class should be similar the
PureAnalogPin but adding a type (Analog or PureAnalog) and should be used
to instantiate the class for A_A0,.. A_A13 using A0, A1... or A8.get()...
(N.B. In ArduinoCore-mbed/variants/GIGA/pins_arduino.h: we found, ... const
uint8_t A0 = PIN_A0... which makes it difficult to replace)
When creating an instance of AdvanceAdc, the constructor will receive a
list object and then use the right approach (code) to get the PinName by
checking the type of the object. The constructor can then use
analogPinToPinName(p.get()) for the classic AnalogPin or use ???
g_pureAAnalogPinDescription[p.get()].name for the PureAnalogPin to get the
PinName.

What do you think?

Still have to check what is that g_pureAAnalogPinDescription...

Regards,

Ben

leonardocavagnis, 2025-01-07:
> I still have to check what is that g_pureAAnalogPinDescription... (i have found the extern declaration in Arduino.h but I don't find where this is set...)

[Here](https://github.com/arduino/ArduinoCore-mbed/blob/f7fee484369904ad55e8c688887f5d8272dd23eb/variants/GIGA/variant.cpp#L19)!