arduino-libraries/Arduino_AdvancedAnalog #83: The example ADC_To_DAC.ino does not work , must copy entire buffeer before write it

PantalaLabs, 2024-12-25:
`
#include <Arduino_AdvancedAnalog.h>
AdvancedADC channelA0(A0);
AdvancedDAC dac0(A12);
void setup()
{
    if (!channelA0.begin(AN_RESOLUTION_12, 96000, 32, 32))
    {
        Serial.println("ADC fail...");
        while (1)
            ;
    }
    if (!dac0.begin(AN_RESOLUTION_12, 96000, 32, 32))
    {
        Serial.println("Failed to initialize DAC 1");
        while (1)
            ;
    }
}

void loop()
{

    if (channelA0.available())
    {
        SampleBuffer inBuffer = channelA0.read();
        SampleBuffer outBuffer = dac0.dequeue();

        for (int i = 0; i < inBuffer.size(); i++)
        {
            outBuffer.data()[i] = inBuffer[i];
        }

        // Write contents of the output buffer to the DACs
        dac0.write(outBuffer);

        // Release the input buffer
        inBuffer.release();
    }
}
`

PantalaLabs, 2024-12-25:
sorry for the formatting, but I pasted the entire code inside the code tag and it didn't work, only the IF block

leonardocavagnis, 2025-01-07:
Can you explain a bit more about what's not working? 
The library is designed like this, you need to copy the buffer before writing it to the DAC.

PantalaLabs, 2025-01-07:
Hi Leonardo 

in the  ADC_To_DAC.ino example there is only the input buffer being passed as an argument to the output buffer.
This way the example doesn't work.

Thank you.

Gibran

```
void loop() {
    if (adc1.available()) {
        dac1.write(adc1.read());
    }
}
```

iabdalkader, 2025-01-10:
Yes, that's one of the earliest examples and it's definitely wrong. You should only call `dac1.write` with a buffer obtained from `dac.dequeue`, and not mix buffers. Nothing bad should happen if you do, the DAC will just not output anything.

closed 2025-01-10 by leonardocavagnis