arduino-libraries/Arduino_AdvancedAnalog #84: What are these pins in the I2S_DAC_Output.ino example? I couldn't find some of them in the GIGA documentation

PantalaLabs, 2024-12-25:
`
AdvancedDAC dac1(A12);
// WS, CK, SDI, SDO, MCK
AdvancedI2S i2s(PG_10, PG_11, PG_9, PB_5, PC_4);
`
Pinout 
https://docs.arduino.cc/resources/pinouts/ABX00063-full-pinout.pdf

Thank You

PantalaLabs, 2024-12-25:
sorry, again, I pasted the code inside the code tag and it didn't format

leonardocavagnis, 2025-01-07:
Hi @PantalaLabs,
The pins are sparsely distributed on the board. Please take a closer look at the documentation!

[image]
[image]
[image]
[image]

closed 2025-01-07 by leonardocavagnis

PantalaLabs, 2025-01-07:
Hi Leonardo , 

Very sorry , my fault.
I just printed the first page of the complete PDF [https://docs.arduino.cc/resources/pinouts/ABX00063-full-pinout.pdf](https://docs.arduino.cc/resources/pinouts/ABX00063-full-pinout.pdf)

Thank you.

Gibran