arduino-libraries/Arduino_AdvancedAnalog #87: nonsense values in example Arduino_AdvancedAnalog/examples/Advanced/ADC_Dual_Mode /ADC_Dual_Mode.ino

StefanL38, 2025-01-12:
The example
[examples/Advanced/ADC_Dual_Mode)
 /ADC_Dual_Mode.ino](https://github.com/arduino-libraries/Arduino_AdvancedAnalog/blob/main/examples/Advanced/ADC_Dual_Mode/ADC_Dual_Mode.ino)

has nonsense values
at a baudrate of 9600
`void setup() {
    Serial.begin(9600);
`
aquiring new values once every **single** millisecond
and updating after the serial printing

`
        // Process the buffer.
        if (millis() - last_millis > 1) {
            Serial.println(buf1.timestamp());  // Print buffer timestamp
            Serial.println(buf1[0]);           // Print sample from first channel
            Serial.println(buf1[1]);           // Print sample from second channel
            
            Serial.println(buf2.timestamp());  // Print buffer timestamp
            Serial.println(buf2[0]);           // Print sample from first channel
            Serial.println(buf2[1]);           // Print sample from second channel

            last_millis = millis();
        }
`

makes no sence.
print so many characters at 9600 baud will take longer than a millisecond 
and this results in a update-rate way below 1 ms because the updating is done **after** the serial printing

If this is somehow different and works add comments to the code that explain why it works

leonardocavagnis, 2025-01-13:
Hey! Thanks for reporting!
Please, open a PR proposing a suitable delay time
Thanks

iabdalkader, 2025-01-13:
> configured to process the buffer every millisecond:

No, it's not. It prints the values if at least 1 millisecond has passed since the last print, to throttle the output. This example is just meant to demonstrate how to configure dual ADCs, you should not be printing ADC values anyway.